dkjonas/Wavin-AHC-9000-mqtt #1: Question: Use of "ttl to rs485"-circuit without RE/DE-pin?

niklascp, 2018-10-04:
Hi. First of all thanks for sharing your awesome work! I was also planning to integrate my AHC-9000 and HA, so this comes very convenient!

This is a hardware question more than a issue. I can see from your schematic that D1 is used as RE/DE for the MAX3072E. Now I don't have a MAX3072E/MAX3485 available. But I do have this one, which is based on MAX13487 with "AutoDirection" - so no RE/DE pin:
https://www.amazon.de/DSD-TECH-SH-U12-MAX13487-Raspberry/dp/B07B667STP/ref=sr_1_3?ie=UTF8&qid=1538679557&sr=8-3&keywords=ttl+rs485

My question is, do you think this will be compatiple with your code, and can be wired just ignoring the use of the D1 pin?

PS: I would put a 5/3.3V level converter between the NodeMCU and the RS485 to TTL since it is 5V.

Thanks!

Br.
Niklas

dkjonas, 2018-10-08:
The communication with the AHC is half duplex, ie. my code sets the direction of the MAX3072E to drive the bus, sends a message, changes the direction to listen on the bus, and then receive a reply message from the AHC. 
I'm not familiar with MAX13487, but from my understanding of the datasheet I think it will work out of the box just ignoring D1. The only thing I'm not quite sure about is what level the TX pin of the ESP is at, when it is not transmitting. The 13487 requires it to be high when listening on the bus. If it is not, then it will be a very small change to the code to make it high when the ESP is not transmitting.

niklascp, 2018-10-27:
Thanks for the response! Finally got time to try it out, and it is working just fine ignoring D1 :)

jascdk, 2018-11-05:
> Thanks for the response! Finally got time to try it out, and it is working just fine ignoring D1 :)

Hi Niklas. How did you wire your interface?

exetico, 2023-02-28:
Now that RS485 is implemented on some cool boards like [this](https://www.aliexpress.com/item/1005002010789649.html), could you please prove a few more details, @niklascp . I know this is a old issue.

So... That brings us to:
Did you end up with this solution, and does it still works as expected? :) 

I've ordered the board I linked to, and hope that it will work just fine.