dkjonas/Wavin-AHC-9000-mqtt #7: The interface is not working:(

jascdk, 2018-11-13:
Hi. I just got the IC MAX3072E with the mail. I tried to attach it to a breadboard with a Wemos D1 mini (instead of a nodemcu - but that should not be a problem). 

After I connected the whole thing the system (subscribing the heat/# topic) it just say true / false all the time.

I am unsure if the diagram here on github is correct due to the official MAX3072E https://www.maximintegrated.com/en/products/interface/transceivers/MAX3072E.html

the diagram looks like this : 
[image]

What could I be doing wrong?

dkjonas, 2018-11-13:
Hi,
The schematic is correct and shows the same pin numbers for the ic as in the datasheet. But it is drawn mirrored.
You should connect it, so that pin 1 (RO) is connected to D7 on the Wemos etc.
Which topic is changing between true and false? /online?

dkjonas, 2018-11-13:
Also, since you are not using nodemcu, you should change your board in platformio.ini to macth the board you are using (should be on of d1, d1_mini, d1_mini_pro, d1_mini_lite depending on exactly which board you have)

jascdk, 2018-11-14:
Hi :) thanks for the quick reply ! I will try to change the board in PlatformIO . I think that could be the problem - I will get back to tell about the results :) thanks again

jascdk, 2018-11-14:
Hi again - regarding the true false topic is that it is the online topic, that is going forth and back :/

dkjonas, 2018-11-14:
The online topic should only be changing when connection to the mqtt server is lost/restored. How often does it change?

jascdk, 2018-11-14:
Hi. It changes every half a minute or so:( I tried define d1_mini in both the “env” and in the board - did not help :( the way I power it for now is via usb. I have only connected the ground , the A and B to the IC. But that should not be an issue I think ?

dkjonas, 2018-11-14:
Strange... it should stay connected even if the AHC is not connected. You can try to remove the ic, so that  only the Wemos is connected to usb, and see if that makes a difference (not very likely).
What mqtt server are you using? Half a minute sounds like it is the server, that drops the connection. There is no 30s delay in my code. At most 16s if the the AHC is not found, but even in that case, it should remain connected.

jascdk, 2018-11-14:
Very strange yes . All my other things connected to the server is running fine. I’m using mosquito through home assistant - maybe I should try to use the native MQTT server in home assistant instead . Which one are you using ?

dkjonas, 2018-11-14:
Ok, I just tried uploading my code to a new nodemcu not connected to anything. It shows the same behaviour as yours (online=true, ~30seconds, then false, then no updates), so that is expected. Not sure why - I will look at it later. I use mosquitto too, so that is not at problem.
But for now I think you should focus on connecting the IC correctly.

jascdk, 2018-11-14:
Hi again:) Nice to know, that Im not the only one having this problem:) I will look forward to an update:)

jascdk, 2018-11-14:
Oh btw. Lets say I don't want to use username and password for the MQTT Broker. What do I have to change in the code then?

dkjonas, 2018-11-14:
Try setting the username/password to "" or to String(NULL) (no quotes).

If that it not working remove "MQTT_USER.c_str(), MQTT_PASS.c_str()," from line 239 in main.cpp.

dkjonas, 2018-11-16:
The reason the connection with the server is lost and reconnected, is that there is a default timeout of 15s in the mqtt library. Within these 15s, the server must be notified that the connection is still alive.
My code polls each of the 16 channels of the AHC for status with a timeout of 1s. So, when the AHC is not connected, the polling will take 16s, and within this time the connection wasn't kept alive. 

I changed the code, so it will now keep the connection alive, even when the AHC is not connected.

dkjonas, 2018-11-16:
Btw. If you test connecting without username/password, please let me know which of the above methods works, so that I can add a comment about it in the code.

dkjonas, 2018-11-16:
It is hard to debug this remotely. You can try uploading a picture of your breadboard. Maybe I can spot if something is wrong.
Do you have any equipment for measuring the signals?

jascdk, 2018-11-16:
I know:) I will take a picture and send you! Yes - I have an old oscilloscope, and multimeter (that's all) . I will start with the picture

jascdk, 2018-11-16:
[image]
[image]

jascdk, 2018-11-16:
It is the best pictures I can get:) Don mind the D1 Mini - It was for another project:)

dkjonas, 2018-11-16:
Are you sure about the patch cable connections? Normally brown is pin 8 and not pin 1.

jascdk, 2018-11-16:
If I hold it as on your diagram it should be the correct one as I see it. Will take a photo of the connection:)

jascdk, 2018-11-16:
[image]

jascdk, 2018-11-16:
It is because I have looked at it from the top as on your diagram - but I think I should have followed the PIN numbers given on the web instead:/ Hope I not have fried something uuuughhh:/

dkjonas, 2018-11-16:
Yep. Pin 1 is orange-white, pin 3 green-white and pin 6 green. So you should also swap green/green-white.
The symbol in the schematic is a standard symbol from the eagle layout editor. Always check the pin numbers :-)

Don't think anything is fried from this connection error.

jascdk, 2018-11-18:
Hi again:) Now I got the connections correctly and guess what - it works - finally. Sorry for all my questions - it was me, who did not have the connections right :D

dkjonas, 2018-11-18:
Great. Always nice to know when something you made is actually working.

I have updated the schematic with the wire colors and no mirrored parts. Hopefully it will make it easier to understand for others.

closed 2018-11-18 by dkjonas