dkjonas/Wavin-AHC-9000-mqtt #9: Best way to trace if it's not working

projectraam, 2019-03-21:
I now have the board and everything seems to be connected as described. It's connecting to the wifi and I can ping the IP.

First time it was connected it also showed up in the logfile for the MQTT broker, but all 16 (number of sensors) request contained strange symbols such as '�'. Then I disconnected the board, restarted the MQTT broker and then reconnected again, now it does not connect to the broker but still connects to the wifi. 

I have also tried to flash the nocemcu again. 

What is the best way to trace where the problem is?

Note. The logs was cleared on restart so cant provide the full log.

projectraam, 2019-03-21:
I am using homeassistant and the plugin "MQTT Server & Web client", the config for that looks like this:

{
  "log_level": "info",
  "certfile": "fullchain.pem",
  "keyfile": "privkey.pem",
  "web": {
    "enabled": true,
    "ssl": false
  },
  "broker": {
    "enabled": true,
    "enable_ws": false,
    "enable_mqtt": true,
    "enable_ws_ssl": false,
    "enable_mqtt_ssl": false,
    "allow_anonymous": false
  },
  "mqttusers": [
    {
      "username": "username",
      "password": "password",
      "readonly": false,
      "topics": [
        "#"
      ]
    }
  ]
}

dkjonas, 2019-03-21:
I don't know the "MQTT Server & Web client" component. I use mosquitto as my broker, so I do not know if the config above is correct.
Can you connect to mqtt using another client - just to make sure the server is working?

projectraam, 2019-03-21:
I am able to connect to the MQTT server via another client and it shows up in the logs

`1553197105: New connection from 192.168.110.32 on port 1883.
1553197105: New client connected from 192.168.110.32`

I have some Sonoff switches connected to my MQTT server and they connect to hassio.local:1883
In my PrivateConfig.h I have the following:
`const String   MQTT_SERVER = "hassio.local:1883"; // mqtt server`

dkjonas, 2019-03-21:
Try without the port number in the address:
const String MQTT_SERVER = "hassio.local"; // mqtt server

Port is defined in main.cpp:
const uint16_t MQTT_PORT = 1883; // mqtt port

projectraam, 2019-03-21:
That did the job, thanks a lot for quick help 👍

closed 2019-03-21 by projectraam