dkjonas/Wavin-AHC-9000-mqtt #10: Socket error

Mugge66, 2019-04-27:
Hey And sry for asking. Im trying to get this to work but it keeps saying socket error disconnecting.

Im thinking it could be related to user/password. 

Im running my mqtt in docker on synology and im not using user or password. 

in the private.sh config file in your script what should the user and password text be. do i just delete you text and leave it like    ""   // username   <-- like this ?

dkjonas, 2019-05-01:
Try setting the username/password to "" or to String(NULL) (no quotes).

If that it not working remove "MQTT_USER.c_str(), MQTT_PASS.c_str()," from line 239 in main.cpp.

Mugge66, 2019-05-08:
"" Works. :-)   Nice Job. :-)

commit 2c8f1e62 (2019-05-11): Added comment about empty username/password

Fix #10
closed 2019-05-11 by dkjonas