dkjonas/Wavin-AHC-9000-mqtt #12: Support for OTA?

jascdk, 2019-05-17:
Hi would it be possible to add OTA (over the air updates)  support to your code? It would be a nice feature if the ESP is tucked away or mounted in a box.

I dont know if it is possible to do OTA  with platformio?

Regards

/Jacob

dkjonas, 2019-05-19:
I don't expect to add any new features - controlling the temperature is the only thing the AHC can do - so there is not much to add. Therefore I don't see the need to be able to update the code once it is installed. In other words, I won't write the code.

But I don't think there are any technical reasons why I can't be included. So if someone creates a PR with the needed code changes, I will try to include it.

closed 2019-05-19 by dkjonas

jascdk, 2019-05-20:
@dkjonas That is understandable :) Thanks for the reply - I will try ask someone to make an PR if they also have an interest in the code. Great code by the way!