dkjonas/Wavin-AHC-9000-mqtt #14: Zero readings and different readings with same ID0

Deez73, 2019-06-28:
Really happy with my AHC 9000 MQTT setup, but I have two different issues with the readings.

Attached is an image illustrating the two problems (MQTT sub dump from my installation)

1. I get several zero readings - First green box from ID2 ... second green box shows same ID2 but with correct values read

2. I get different readings with the same ID0 (red boxes)... Clearly it's from different sources, but the ID gets the same number (ID0) so the readings get mixed up

My installation is with 18 floorheating sources.

[image]

dkjonas, 2019-06-29:
How often do you get zero readings? Is it only happening on one specific channel? It may be because the thermostat looses connection with AHC. My code checks if a thermostat is connected to the controller before reading temperature and battery level, but there may be a delay from this check until it reads the actual values, and maybe connection is lost during this delay. You can try to move line 333 to 352 in main.cpp up to line 282. This should limit the delay.

As for the mixed up values: Do you have several outputs connected to one thermostat? I never tested this, so there may be a bug in handling this. Only ID0 has this problem?

Can you add timestamps to the log above? It is a bit hard to see how often this happens.

Deez73, 2019-06-30:
Zero readings happen all the time, the example in the screenshot is readings from just 3 minutes or so. Zero readings happen on all ID’s/channels. They happen many times a day.

The mixed up values are only happening on ID0 and it thus looks like several thermostats report in as ID0 (or the MQTT message reports them as ID0)

I have only one output assigned to each thermostat.

dkjonas, 2019-06-30:
Ok, haven't seen/heard of this before. Wondering if it is due to the high number of thermostats. I only have 5 in my setup. 
What happens if you set NUMBER_OF_CHANNELS in WavinController.h to something lower ie. 5?

Does all channels report correct values apart from these two errors?

Is the source for the wrong values for channel 0 always the same?

Do you have a display connected to the AHC?

Deez73, 2019-06-30:
No display connected, but I have two AHC9000 because of the 18 channels. Have now tried disconnecting the second AHC9000 to test with the primary one with 16 channels in use. I have a display, but not using it (not connected) after building and using your interface.

When channels report a value (not zero), the value seems correct.

Think it is 3-4 channels that incorrectly report as ID0 and then gets mixed up.

I will do some further testing and logging values to see if I can isolate causes. Will also try to do the source adjusting you suggest (including limiting number of channels) to test, but it will take some days recompiling and retesting. Will report back here as soon as I have more information. Let me know if you have further ideas.

dkjonas, 2019-06-30:
My code is not handling multiple AHC's. Could very well be the source of the problem.

Deez73, 2019-07-01:
You are correct! After disconnecting my second AHC 9000 controller, the issues are gone (no errors whatsoever since last night). Thank you very much for your help, I can live without the last few channels and is closing this issue since it was a problem on my side, not with your code.

closed 2019-07-01 by Deez73