dkjonas/Wavin-AHC-9000-mqtt #15: Wavin AHC 9000 Display Collision?

joak0068, 2019-08-16:
Hi dkjonas,

Thank you for the great work you have been doing. I was able to connect to my wavin controller, and read out all data on my mqtt server - awesome!

However I encountered an issue when connecting the ESP.
I have a Display connected to my AHC9000 - which will beep from time to time when ESP is connected, have you seen this issue?

Seems to happen on every poll, but also a bit random. Maybe there's some kind of collision between this display and the ESP?

They are both connected with a RJ45 cable.

Do you have any idea if this can be solved without disconnecting one of them? Any pointers would be great for me to start digging.

Display:
https://www.billigvvs.dk/Varmesystemer-Gulvvarme-Wavin--Wavin-AHC-9000-display-m-trykfoelsom-skaerm-718250.html

BR Joakim

jascdk, 2019-08-16:
Hi Joakim :) i Can give you a quick answer - the project does not work when the display is connected . Try disconnect the display and only use this esp project connected instead . The problem is that it shares the connection .

joak0068, 2019-08-16:
Yes, I was thinking if it was colliding somehow. Thank you for the quick answer :) I'll make do without the display then.

closed 2019-08-16 by joak0068

jascdk, 2019-08-16:
Did you work it out :) ?