dkjonas/Wavin-AHC-9000-mqtt #16: Support for Wavin-AHC-9000 2-zone control unit?

nens90, 2019-10-09:
Hey Jonas

This is a very cool project and i'm eagerly trying to get it to work. I have followed all the instructions and looked through all the reported issues to compare, but still no luck.

I have tried to scope the output of the MAX3485 and the signal looks fine. Then I came to wonder, since you are mentioning that there is 16 channels on the Wavin AHC-9000/Jablotron AC-116, but I am trying to use the 2-zone version.

Do you know if the 2-zone Wavin AHC-9000 control unit is compatible with your implementation ? 

Kind regards
Jens Nielsen

nens90, 2019-10-09:
Thanks for the fast reply =)
Yah, okay. It looks like form the installation instructions, that the RJ-45 connector is a service port. Whatever that means.

[image]
[Wavin AHC 9000 styreenhed til gulvvarme - PDF.pdf](https://github.com/dkjonas/Wavin-AHC-9000-mqtt/files/3708506/Wavin.AHC.9000.styreenhed.til.gulvvarme.-.PDF.pdf)

dkjonas, 2019-10-09:
Not much information in the pdf. The circuit looks like the rj45 connector could be rs422 like the 16ch version, so maybe there is hope.
You can try to ask Wavin for documentation on the protocol. They responded quickly when I asked for the 16ch docs.

nens90, 2019-10-09:
Yes, the transiver connected to the RJ-45 is RS422. I will try and ask Wavin.
Thanks for your support =)

closed 2019-10-09 by nens90

jmork, 2022-11-06:
@nens90 I wonder, did you ever get things working with the 2ch version, or get an answer from Wavin? I would like to do the same thing...

stefboerrigterWavin, 2022-11-07:
I would say that the 2ch works similar to the 16 channel variant with respect to modbus commands.

tiede, 2024-09-25:
@jmork / @nens90 Did you get information regarding the protocol used for the 2-channel unit?

jmork, 2024-09-25:
@tiede No, unfortunately not.