dkjonas/Wavin-AHC-9000-mqtt #19: Support IR sensor on thermostat.

beermann, 2019-12-18:
Hi, 

Great job on making the AHC9000 work with HomeAssistant, is there any change  of you adding support for the IR temperatur sensor? This would give feedback of the floor temperatur.

dkjonas, 2019-12-18:
This should be trivial to add, but I don't have any hardware to test it.

Can you please try make the following change in WavinController.h:
Change 
`static const uint8_t ELEMENTS_AIR_TEMPERATURE = 0x04;`
to
`static const uint8_t ELEMENTS_AIR_TEMPERATURE = 0x05;`

Now I would expect it to return the floor temperature instead of the air temperature on all thermostats. If this works, then I will make the necessary changes to have it return both values.
If you have a thermostat without IR, then the temperature will most likely be read as 3276.7 - please confirm this if possible.

beermann, 2019-12-28:
Hi,

I have just testet your suggested change, It now shows the temperature of
the IR, but the onces that do not have an IR just shows 0. I du not se a
reading of 3276.7 is that expected to be shown in the MQTT topic?
I have attached the output from MQTT with the IR change, its the thermostat
1 that has IR.

Do you know what would happen if I connect 2 AHC9000 units with this ESP?
will it work? ( i do not have the link in place yet, but i do have 2
AHC-9000 units that i plan to connect.)

Sune

heat/floor5CCF7FC05F21/online False
homeassistant/climate/floorECFABC3944CC/0/config {"name": "floorECFABC3944CC_0_climate", "current_temperature_topic": "heat/floorECFABC3944CC/0/current", "temperature_command_topic": "heat/floorECFABC3944CC/0/target_set", "temperature_state_topic": "heat/floorECFABC3944CC/0/target", "mode_command_topic": "heat/floorECFABC3944CC/0/mode_set", "mode_state_topic": "heat/floorECFABC3944CC/0/mode", "modes": ["heat", "off"], "availability_topic": "heat/floorECFABC3944CC/online", "payload_available": "True", "payload_not_available": "False", "qos": "0"}
homeassistant/climate/floorECFABC3944CC/1/config {"name": "floorECFABC3944CC_1_climate", "current_temperature_topic": "heat/floorECFABC3944CC/1/current", "temperature_command_topic": "heat/floorECFABC3944CC/1/target_set", "temperature_state_topic": "heat/floorECFABC3944CC/1/target", "mode_command_topic": "heat/floorECFABC3944CC/1/mode_set", "mode_state_topic": "heat/floorECFABC3944CC/1/mode", "modes": ["heat", "off"], "availability_topic": "heat/floorECFABC3944CC/online", "payload_available": "True", "payload_not_available": "False", "qos": "0"}
homeassistant/climate/floorECFABC3944CC/2/config {"name": "floorECFABC3944CC_2_climate", "current_temperature_topic": "heat/floorECFABC3944CC/2/current", "temperature_command_topic": "heat/floorECFABC3944CC/2/target_set", "temperature_state_topic": "heat/floorECFABC3944CC/2/target", "mode_command_topic": "heat/floorECFABC3944CC/2/mode_set", "mode_state_topic": "heat/floorECFABC3944CC/2/mode", "modes": ["heat", "off"], "availability_topic": "heat/floorECFABC3944CC/online", "payload_available": "True", "payload_not_available": "False", "qos": "0"}
homeassistant/climate/floorECFABC3944CC/3/config {"name": "floorECFABC3944CC_3_climate", "current_temperature_topic": "heat/floorECFABC3944CC/3/current", "temperature_command_topic": "heat/floorECFABC3944CC/3/target_set", "temperature_state_topic": "heat/floorECFABC3944CC/3/target", "mode_command_topic": "heat/floorECFABC3944CC/3/mode_set", "mode_state_topic": "heat/floorECFABC3944CC/3/mode", "modes": ["heat", "off"], "availability_topic": "heat/floorECFABC3944CC/online", "payload_available": "True", "payload_not_available": "False", "qos": "0"}
homeassistant/climate/floorECFABC3944CC/4/config {"name": "floorECFABC3944CC_4_climate", "current_temperature_topic": "heat/floorECFABC3944CC/4/current", "temperature_command_topic": "heat/floorECFABC3944CC/4/target_set", "temperature_state_topic": "heat/floorECFABC3944CC/4/target", "mode_command_topic": "heat/floorECFABC3944CC/4/mode_set", "mode_state_topic": "heat/floorECFABC3944CC/4/mode", "modes": ["heat", "off"], "availability_topic": "heat/floorECFABC3944CC/online", "payload_available": "True", "payload_not_available": "False", "qos": "0"}
homeassistant/climate/floorECFABC3944CC/5/config {"name": "floorECFABC3944CC_5_climate", "current_temperature_topic": "heat/floorECFABC3944CC/5/current", "temperature_command_topic": "heat/floorECFABC3944CC/5/target_set", "temperature_state_topic": "heat/floorECFABC3944CC/5/target", "mode_command_topic": "heat/floorECFABC3944CC/5/mode_set", "mode_state_topic": "heat/floorECFABC3944CC/5/mode", "modes": ["heat", "off"], "availability_topic": "heat/floorECFABC3944CC/online", "payload_available": "True", "payload_not_available": "False", "qos": "0"}
homeassistant/climate/floorECFABC3944CC/6/config {"name": "floorECFABC3944CC_6_climate", "current_temperature_topic": "heat/floorECFABC3944CC/6/current", "temperature_command_topic": "heat/floorECFABC3944CC/6/target_set", "temperature_state_topic": "heat/floorECFABC3944CC/6/target", "mode_command_topic": "heat/floorECFABC3944CC/6/mode_set", "mode_state_topic": "heat/floorECFABC3944CC/6/mode", "modes": ["heat", "off"], "availability_topic": "heat/floorECFABC3944CC/online", "payload_available": "True", "payload_not_available": "False", "qos": "0"}
homeassistant/sensor/floorECFABC3944CC/0/config {"name": "floorECFABC3944CC_0_battery", "state_topic": "heat/floorECFABC3944CC/0/battery", "availability_topic": "heat/floorECFABC3944CC/online", "payload_available": "True", "payload_not_available": "False", "device_class": "battery", "unit_of_measurement": "%", "qos": "0"}
homeassistant/sensor/floorECFABC3944CC/1/config {"name": "floorECFABC3944CC_1_battery", "state_topic": "heat/floorECFABC3944CC/1/battery", "availability_topic": "heat/floorECFABC3944CC/online", "payload_available": "True", "payload_not_available": "False", "device_class": "battery", "unit_of_measurement": "%", "qos": "0"}
homeassistant/sensor/floorECFABC3944CC/2/config {"name": "floorECFABC3944CC_2_battery", "state_topic": "heat/floorECFABC3944CC/2/battery", "availability_topic": "heat/floorECFABC3944CC/online", "payload_available": "True", "payload_not_available": "False", "device_class": "battery", "unit_of_measurement": "%", "qos": "0"}
[... cut ...]
homeassistant/sensor/floorECFABC3944CC/1/config {"name": "floorECFABC3944CC_1_battery", "state_topic": "heat/floorECFABC3944CC/1/battery", "availability_topic": "heat/floorECFABC3944CC/online", "payload_available": "True", "payload_not_available": "False", "device_class": "battery", "unit_of_measurement": "%", "qos": "0"}
heat/floorECFABC3944CC/1/target 21.5
heat/floorECFABC3944CC/1/mode heat
heat/floorECFABC3944CC/1/output off
heat/floorECFABC3944CC/1/current 24.0
heat/floorECFABC3944CC/1/battery 60

paller, 2020-01-19:
I also have a IR floor thermostat if you need help testing. If I remember correctly does without IR always report 0 for the floor and I'm not sure there is any other way to detect if a thermostat has IR support or not.

paller, 2020-01-19:
> Do you know what would happen if I connect 2 AHC9000 units with this ESP? will it work? ( i do not have the link in place yet, but i do have 2 AHC-9000 units that i plan to connect.) Sune

There is a initialization protocol for that. The ESP will have to detect and enumerate each unit and use the assigned IDs in the following communication to select either one of the units.
It shouldn't be difficult to implement but I can't test it with a single unit.

Pikfisk, 2020-03-13:
Hi, 
While we are at adding more signals, how about also adding Humidity & dew point temperature to the readings?
Those would be interesting to see as well.

BR Jonas
[image]

osos, 2021-09-19:
Link to specs here: https://groups.google.com/g/pymodbus/c/e2q8aYSIWlI

osos, 2021-09-21:
I made a pull request here: https://github.com/nic6911/Wavin-AHC-9000-mqtt/pull/1

Maybe it could benefit this main project as well.