dkjonas/Wavin-AHC-9000-mqtt #29: Hjælp til udvikling af ESPHome component

heinekmadsen, 2022-02-15:
Hej....

Jeg har igennem noget tid arbejdet på et component til ESPhome til Wavin AHC 9000, hvilken sådanset også er kørende med climate entities i HA og mulighed for at stille temperatur mv.
Det eneste der driller mig er muligheden for at sætte en termostat i standby(off på gamle modeller).
Jeg kan få det til at slå igennem 1 gang, men næste gang componenten opdaterer sætter den termostaten tilbage til on.

Mener ikke jeg havde denne udfordring med mqtt koden du har lavet. Har du mon nogen ide/inputs til hvorfor den opfører sig sådan?

https://github.com/heinekmadsen/esphome_components/tree/main/components/wavinAhc9000

Venlig hilsen
Heine

dkjonas, 2022-02-20:
Som jeg forstår dit spørgsmål, så har du problemer med at termostaten ikke husker indstillingen, næste gang den polles?
Jeg mener ikke at have haft problemer med det, men har aldrig i praksis brugt muligheden for at ændre mode, så det er muligt det heller ikke har virket i min kode.

heinekmadsen, 2022-02-20:
Jae det kan man nok godt sige at det er sådan... Men hvis jeg slet ikke har koblet noget til modbus på styreenheden, så virker det fint. Altså sætter off på termostaten, og så forbliver den off. Hvis jeg har min esp koblet på med din mqtt kode, så forbliver den også på off.. men har jeg esphome på min esp, så slår den tilbage til on.
Så har isoleret det til at det må være noget med esphome og ikke wavin anlægget.

Jeg kan se at du i din kode sætter en standby temperature som noget af det første. Er ikke sådan super til selve modbus, men kan det være nogle krav til at denne sættes før det virker?

dkjonas, 2022-02-21:
Som jeg husker det, sætter jeg standby temperaturen lav, fordi jeg havde brug for at lave en mode, som slukkede for varmen til brug ved Google Assistant. Assistant har en off mode, som jeg ikke lige kunne se andre måder at lave end ved hjælp af standby med lav temperatur.

Jeg har lige hurtigt kigget din kode igennem, og så vidt jeg kan se, er det eneste du skriver til controlleren target temperatur. Du mangler at skrive mode til PACKED_DATA_CONFIGURATION når epshome skifter mode.

heinekmadsen, 2022-02-22:
Ja det er korrekt jeg ikke har implementeret koden til at sætte standby mode. I første omgang vil jeg bare gerne kunne læse den rette værdi ud.
Dette kan jeg sagtens. Kan se i loggen når jeg skifter standby mode til on på termostaten, at så skifter den fra 0 til 1, men ved næste opdatering er den skiftet tilbage til 0 og termostaten skifter også sin standby mode tilbage til off.

dkjonas, 2022-02-22:
Det har jeg ikke noget bud på. 
Mener som sagt ikke jeg har haft problemer, men har heller aldrig brugt funktionaliteten.

closed 2022-02-24 by dkjonas