dkjonas/Wavin-AHC-9000-mqtt #35: Sporadic fail on readRegisters

marn06, 2023-06-15:
Hi dkjonas

Thanks a lot for the great work you have done with this project. I am using your code with a few small changes on an ESP32-C3. Unfortunately I see sporadic fails when reading registers, is this something you also experience? See mqtt log, where POLL represents the start of reading registers on all channels:
[image]

marn06, 2023-06-15:
Hi @dkjonas 
I actually think I found the solution, at least I haven't had any fails for hours, with the new code in bold added:
    Serial1.begin(38400, SERIAL_8N1, rxPin, txPin);
    **ESP_ERROR_CHECK(uart_set_mode(UART_NUM_1, UART_MODE_RS485_HALF_DUPLEX));**

Roarrk, 2023-11-12:
Hey @marn06 

Would you mind sharing your esp32-c3 repo - and are you able to use uart directly from the esp-c3 or do you still need a max3072e? My setup with a 8266 works, but I'm having intermittent issues so I'm looking to try alternatives

marn06, 2023-11-13:
Hi @Roarrk 
Sure, but my code is based on this board, let me know if you are still interested.
https://www.ustepper.com/shop/#!/products/esphome-modbus-module