dkjonas/Wavin-AHC-9000-mqtt #37: How to find other units information

andreapier, 2024-04-01:
Hi, this is not an issue but I need help nonetheless.
I own a mechanical ventilation system by wavin. A RDCD 3.0 SH to be precise.
It has an RJ45 but it's not the classica LAN port. The device actually turns off if I plug a normal ethernet cable.
I suspect it the same connector described here, and I'd like to know more about it.
Do you have any guidance on how/where I can get more info?
I don't have a manual or a referente to the technician who installed it. Wavin is totally useless too, I tried to contact them but it was a dead end.
Thanks for your time and your effort on this peoject!!

jascdk, 2024-04-01:
Hi :) I have a suggestion for you. This project here has evolved quite a lot since a lot of people these days use Home Assistant managing their smarthomes. Therefore you might take a look at this repo here https://github.com/heinekmadsen/esphome_components/tree/main which has code for controlling a Wavin AHC9000 floor heat controller. Im using a esp32 device controlling modbus on the abc unit - and also the same module for controlling our HVAC system from Nilan.

If you want, I can help you get in contact with this guy who has this repo here - maybe he has some modules left. https://github.com/nic6911/ESP32_Modbus_Module

I think your system may use modbus for communication, but if I were you, I would contact wavin here in Denmark and ash for the modus-protocol on the system you have. Then we can se if we can figure out communicating with it;)

All the best

Jacob / Denmark

stefboerrigterWavin, 2024-04-02:
@andreapier 
I found these registers:
`
Setting	                                || Modbus Address(es)   || Remarks
Set unoccupied air flow	                || 1511 & 1513	        ||
Set economy air flow	                || 1507 & 1509	        ||
Set comfort air flow	                || 1503 & 1505	        ||	
Set State 	                            || 2001	                ||	1: protection (stopped, only if enabled by installer); 2: unoccupied; 3: economy; 4: comfort
Operating time for filter replacement   || 1623	                ||  Installer sets the time until the filter needs to be replaced. After this time, the user should get a notification
Operating time filter	                || 1621	                || How many hours since last filter replacement. If this value exceeds the value set in register 1623 (set by installer), a B-alarm is triggered
General Alarm	                        || 4012                 || (B-alarm), 4010 (A-alarm)	
General Alarm RESET/ACK	                || 3003, 3004	        ||
By-pass on/off	                        || 2028	                || 0: closed (full heat recovery); 1: open (heat recovery bypassed). 
Supply air temperature	                || 505	
Extract air temperature	                || 509	
Exhaust air temperature	                || 513	
Outside air temperature	                || 501	
`

andreapier, 2024-04-02:
> @andreapier I found these registers: [...]

Mini sharingour source?
Btw, THANKS!!! This looks like exactly what I was looking for!

stefboerrigterWavin, 2024-04-03:
I don't believe the sources are re-usable for this project or a relevant project. It's tailored for our products. 
Next to that, we don't have a policy to open our source-code of existing products.

andreapier, 2024-04-03:
> I don't believe the sources are re-usable for this project or a relevant project. It's tailored for our products. Next to that, we don't have a policy to open our source-code of existing products.

Sorry, yesterday I was on mobile and did not immediately notice your username... I guess you are a wavin employee.
First, let me say thank you again. Second, I did not mean to share the source code. I meant if you could share the source where you got those registers, like an installer manual or anything like that (probably the word "source" was a little misleading). But I guess it is internal documentation, and it is just fine if company policies disallow to share it.

andreapier, 2024-04-08:
> I don't believe the sources are re-usable for this project or a relevant project. It's tailored for our products. Next to that, we don't have a policy to open our source-code of existing products.

@stefboerrigterWavin Sorry to annoy you, but I was wondering if you could at least confirm that the schematic of the RJ45 port in RDCD 3.0 SH is the same as the AHC-9000 or not. Thanks!

stefboerrigterWavin, 2024-04-09:
No Problem, 
Yes, the information is obtained from internal documents. 
The RJ45 port on the Ventilation unit is suitable for modbus communication (protocol see image i shared before).
Wiring is(should be) identical to other Wavin products for Modbus:
[image]

dkjonas, 2024-04-09:
I don't think the device shares the software protocol of this project, so I will suggest creating a different repo for the device, where you can share the details.

closed 2024-04-09 by dkjonas

andreapier, 2024-04-28:
> I don't think the device shares the software protocol of this project, so I will suggest creating a different repo for the device, where you can share the details.

Done, [here is the repo](https://github.com/andreapier/esp-rdcd) for people who could be interested.